iizukanao/picam-streamer #1: Audio & Video possible in one stream?

ringzer, 2016-02-02:
Hi,

This project looks awesome and exactly what I'm looking for - an easy audio and video streaming application I can install on a raspberry pi.

Can you confirm the steam at http://RASPI_IP/view/ is video and audio, assuming I add a camera and microphone to the Pi? Also what app would you recommend on android to watch this stream? Would VLC work?

Thanks,

Ciaran

iizukanao, 2016-02-03:
Thanks!

> Can you confirm the steam at http://RASPI_IP/view/ is video and audio, assuming I add a camera and microphone to the Pi?

Yes. You need a Raspberry Pi Camera Module, not a USB webcam.

> Also what app would you recommend on android to watch this stream? Would VLC work?

Any app that supports RTSP would work, including VLC. Also, Android's VideoView class natively supports RTSP. In VLC, enter "rtsp://RASPI_IP:80/live/picam" as a URL.

ringzer, 2016-02-10:
Thank you for your quick reply and apologies for the delay in replying.

I have the Pi NoIR Camera and have ordered a USB microphone.

I am facing some issue in getting it setup however. This is what I have done:

1. Downloaded picam-streamer an wrote to a 16GB card.
2. Without camera plugged in, expanded file system and rebooted
3. Plugged in camera and rebooted again
4. The red LED on the camera is lighting up and when I browse to http://192.168.0.12/view I can see "Video stream" in the page title and can see something between <script> when I view source. However I the page is blank.
5. I have tried in Firefox, Chrome and also the rtsp link via VLC.

Another thing I've noticed is that with camera plugged in, the Pi seems to be somewhat unresponsive - looking at top I can see picam using quite a bit of CPU
```
  476 root      20   0  119336  16632   5064 R 21.3 13.8   2:38.23 node
  813 pi         0 -20  102652   6556    696 S 17.1  5.4   2:05.02 picam
   30 root       1 -19       0      0      0 S  4.2  0.0   0:30.60 VCHIQ-0
  954 pi        20   0    5080   1060    724 R  1.3  0.9   0:09.74 top

```
I'm happy to post log files but I'm not sure where they are. i would like to resolve this but I might also try and install the complete picam image.

iizukanao, 2016-02-11:
Please allow a few minutes for the /view/ URL to be ready for use. The boot time depends on whether you are using Raspberry Pi 1 or 2.

That `top` command output looks pretty normal and such load itself will never make the Pi unresponsive. Login to your Pi, `cd picam` then run `touch hooks/start_record` and `touch hooks/stop_record` to check if the camera is capturing correctly or not (see https://github.com/iizukanao/picam#recording for usage). If you use VLC, RTSP URL is rtsp://PI_IP:80/live/picam (replace PI_IP with your Pi's IP).

I think it is more of a hardware issue such as power supply problem rather than software. System-level error messages will appear in /var/log/messages. If you think this is a software problem, please post several lines of `vmstat 10` output when the Pi is recording video.

ringzer, 2016-02-11:
I am getting an error when trying to run those touch commands:

```
pi@raspberrypi ~/picam $ touch hooks/start_record
touch: cannot touch ‘hooks/start_record’: No such file or directory
pi@raspberrypi ~/picam $ sudo touch hooks/start_record
touch: cannot touch ‘hooks/start_record’: No such file or directory
pi@raspberrypi ~/picam $
pi@raspberrypi ~/picam $ which touch
/usr/bin/touch

```
Seems odd that I cant run the touch command given that it is supposed to create a file but it is complaining there isn't such a file and the touch command exists.

I will try and run on a Pi 2 and a different power supply, but I believe they are good power supplies.

iizukanao, 2016-02-12:
Sorry, I was wrong. You need to be in ~/picam-streamer/ directory. Replace `cd picam` with `cd ~/picam-streamer` and try the touch command again. Recorded file will appear in ~/picam-streamer/archive/.

ringzer, 2016-02-24:
Apologies for the delay in replying, but I was waiting for my USB microphone to arrive, which it now has: [see here for details](http://www.banggood.com/USB-Microphone-For-Raspberry-Pi-p-993726.html)

I am getting an audio and video rtsp stream but it is choppy and jumpy due to high CPU utilisation node and picam are almost maxing out the Pi.

I have a Raspberry Pi 1 Model B.

`
top - 22:07:48 up 6 min,  3 users,  load average: 1.46, 1.36, 0.71
Tasks:  89 total,   3 running,  86 sleeping,   0 stopped,   0 zombie
%Cpu(s): 74.5 us, 23.6 sy,  0.0 ni,  0.0 id,  0.0 wa,  0.0 hi,  1.9 si,  0.0 st
KiB Mem:    380072 total,   216788 used,   163284 free,    21092 buffers
KiB Swap:   102396 total,        0 used,   102396 free.   113196 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
  434 root      20   0   86216  25588  13288 R 61.8  6.7   2:15.66 node
  733 pi         0 -20   94864  13328   7600 R 30.3  3.5   1:41.48 picam
   30 root       1 -19       0      0      0 S  5.1  0.0   0:16.88 VCHIQ-0
  842 pi        20   0    5080   2548   2132 R  2.2  0.7   0:01.58 top
    3 root      20   0       0      0      0 S  0.3  0.0   0:00.51 ksoftirqd/0
   19 root      20   0       0      0      0 S  0.3  0.0   0:00.75 kworker/0:1
  518 root      20   0   25172  13072   6988 S  0.3  3.4   0:01.92 Xorg
  710 pi        20   0  102124  21932  18560 S  0.3  5.8   0:04.51 lxpanel
`

When I disconnect the microphone top shows
`
  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
  733 pi         0 -20   98960  17732   7600 S 75.3  4.7   5:16.10 picam
  434 root      20   0   88940  27064  13288 R 21.7  7.1   7:31.09 node
  842 pi        20   0    5080   2552   2132 R  1.6  0.7   0:12.39 top
`
When I reboot with the microphone still disconnected the video stream works perfectly fine with top showing
`
top - 22:27:30 up 6 min,  3 users,  load average: 0.90, 0.75, 0.42
Tasks:  89 total,   2 running,  87 sleeping,   0 stopped,   0 zombie
%Cpu(s): 54.7 us, 19.4 sy,  0.0 ni, 23.2 id,  0.0 wa,  0.0 hi,  2.8 si,  0.0 st
KiB Mem:    380072 total,   212144 used,   167928 free,    19144 buffers
KiB Swap:   102396 total,        0 used,   102396 free.   112388 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
  429 root      20   0   85824  24996  13296 R 53.9  6.6   1:33.87 node
  651 pi         0 -20  102652  13044   7196 S 16.8  3.4   1:04.65 picam
   30 root       1 -19       0      0      0 S  4.2  0.0   0:15.83 VCHIQ-0
  769 pi        20   0    5080   2536   2120 R  1.6  0.7   0:00.25 top
    7 root      20   0       0      0      0 S  0.3  0.0   0:00.44 rcu_preempt
`

So it seems that there is either an issue with the USB microhphone I have or with running picam with a video and microphone on the Pi 1 B

Interestingly the total CPU % is about 71% without the microphone and about 92% with the microphone and 20% increase for stream audio doesn't seem terrible compared to the base level of 71% for video. It seems to me that the Pi 1 B isn't powerful enough. 

I can try running picam-streamer on a Pi 2 in the next few days but I'm curious to know what you think the issue is. 

Thanks

iizukanao, 2016-02-25:
Hmm, that's strange.
When I open /view/ with a browser, Pi 1 Model B's load is:

* node: 50% CPU
* picam: 35% CPU
* load average: 1.5

It's undoubtedly heavy, but both audio and video are smooth.

Pi 2 is very powerful compared to Pi 1 and you should use it if you can.

Also, if you're using WiFi dongle, check if your Pi's network bandwidth is sufficient. picam produces 2Mbps video + 40kbps audio by default.

ringzer, 2016-03-09:
Apologies for the delay in replying.

To solve my problem I can either get a new microphone or a Pi 2 but given that I already have a Pi I would prefer to try another microphone.

Can you recommend a USB microphone? Are there any particular requirements for the USB microphone to work with Picam?

Also, the Pi is connected to the network by ethernet so bandwidth shouldn't be an issue.

iizukanao, 2016-03-10:
Could you please confirm that the following command records 30 seconds of audio without problems?

    $ /usr/local/bin/ffmpeg -f alsa -ac 1 -i hw:1,0 -c:a libfdk_aac -ar 48000 -ab 40000 -t 30 out.m4a

The microphone I'm using is this one.
https://github.com/iizukanao/picam#recommended-hardware
However, this product may not be readily available outside of Japan. I think any popular cheap USB microphone will work.

ringzer, 2016-03-10:
Let me try that command and get back to you.

As for the microphone, the one I have this this one: http://www.banggood.com/USB-Microphone-For-Raspberry-Pi-p-993726.html which is a microphone with a USB connection.

The configuration you have is a USB to 3.5mm adapter and a 3.5mm microphone. I wonder is that the required setup, rather than a microphone which plugs directly into USB.

iizukanao, 2016-03-10:
> The configuration you have is a USB to 3.5mm adapter and a 3.5mm microphone. I wonder is that the required setup, rather than a microphone which plugs directly into USB.

That is irrelevant from a standpoint of software. Any USB microphone should work. I used standalone USB microphone without adapter in the past and it worked.

sakshirawal14, 2018-03-09:
Hi
I'm trying to use the picam library. When I try to run picam with the following steps:
$ cd picam
$ ./picam --alsadev hw:1,0

I get the following error:
error creating state dir (./state): file exists

Unable to proceed further. I tried running the "touch hooks/start_record" command too, but it says 
pi@raspberrypi ~/picam $ touch hooks/start_record
touch: cannot touch ‘hooks/start_record’: No such file or directory
pi@raspberrypi ~/picam $ sudo touch hooks/start_record
touch: cannot touch ‘hooks/start_record’: No such file or directory
pi@raspberrypi ~/picam $
pi@raspberrypi ~/picam $ which touch
/usr/bin/touch

If anyone could provide pointers to resolve this issue, it would be highly appreciated.
Thanks in advance. Shall be highly grateful

iizukanao, 2018-03-11:
@sakshirawal14 Please create new issue if you still have problems.

closed 2018-03-11 by iizukanao