iizukanao/picam-streamer #4: For school....

ArchGalileu, 2016-10-24:
Hello iizukanao

How to enable HLS because mobile ;)

Best regards
ArchGalileu

iizukanao, 2016-11-07:
@ArchGalileu Hi, sorry for the late reply!

Login via SSH to your Pi, then edit /home/pi/picam-streamer/lib/server.js line 44 and add `--hlsdir` and `/home/pi/public/hls` to the command line options. The result will look like:
```js
picam = spawn('nice', ['-n', '-20', 'ionice', '-c1', '-n0', 'sudo', '-u', "#" + PICAM_UID, PICAM_PATH, '--rtspout', '--alsadev', 'hw:1,0', '--rotation', '180', '--hlsdir', '/home/pi/public/hls'], {
```

Then, reboot your Pi. After reboot, you can access HLS at http://YOUR_PI_IP_ADDRESS/hls/index.m3u8

ArchGalileu, 2016-11-08:
@iizukanao Thank you me and the kids apreciate ;)

Best regards

iizukanao, 2016-11-08:
@ArchGalileu You're welcome. Please let me know if you have any questions.

closed 2016-11-08 by iizukanao