iizukanao/picam-streamer #6: not working

todoistobe, 2016-12-18:
i dd-ed the 2016-09-05-picam-streamer-1.4.3.img onto an sd card and it boots just fine. i can ssh into the pi. but nothing at http://raspberrypi.local/view/ and neither node nor picam are running. 

what am i doing wrong?

iizukanao, 2016-12-18:
@todoistobe Did you check MD5 or SHA256 sums of the downloaded file?

If it's OK, is your Raspberry Pi Camera properly connected? To see if it's working, SSH into your Pi and stop the service with `sudo service picam-streamer stop`, then try using `raspivid`.

todoistobe, 2016-12-18:
okay, i found the problem: camera connection. i disconnected and reconnected the camera module and now it works like a charm!

thanks!

closed 2016-12-18 by todoistobe