iizukanao/picam-streamer #7: Is it possible to change the default audio and video settings?

MathewH, 2017-02-18:
Hi,

I have been using your project with great success so far. The only I can't do yet is change the default settings or use any commands for the picam. Is it possible, with this project, to control the picam directly or at least via settings?

Thanks,
Mathew

ErHaMep, 2018-11-10:
Hi,

May be you can read this issue. It contains an example of changing the picam settings.
[https://github.com/iizukanao/picam-streamer/issues/3]
In line 44 you can add or change your settings of picam.
I changed the image size in this way.

Succes,
Er

MathewH, 2019-01-22:
Hi,

Thanks for the response to my problem. I managed to change the settings to my liking, altough using 1080p resolution sometimes has unexpected issues. But that is more likely due to the limit of the hardware I guess.

Thanks,
Mathew

closed 2019-01-22 by MathewH