iizukanao/picam-streamer #9: 4 channel microphone

MichaelAronsen, 2017-05-18:
Hi,

Nice job on this project, works very well.

Only issue I have is, I use a ps3 eye webcam for microphone, as it has a 4 channel microphone array, giving very good sensitivity (baby cam, good to hear every little squeek :)).

But this does not work with picam, any possibility of getting it to work?

iizukanao, 2017-05-19:
Unfortunately, picam-streamer does not support USB cameras.

iizukanao, 2017-05-19:
@MichaelAronsen Can you record that audio with `arecord` command?

MichaelAronsen, 2017-05-19:
Right, so i'm using the pi camera module for video.

I can record audio from the webcam with arecord, yes.

4 channels, 16000Hz and SB16_LE format.

iizukanao, 2017-05-19:
@MichaelAronsen Oh, that's good to know. I think I can add support for 4-channel audio. Does ps3 eye microphone also support 2-channel audio? If yes, is 2-channel already working with picam?

TomUser1, 2017-11-27:
whether there can be a problem in it?  : 
"$arecord -D mic_sv  -r 44100 -f S16_LE -t wav -d 0 -c 2  test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
arecord: set_params:1233: Sample format non available
Available formats:
- S32_LE"

Сan you add support of the S32_LE FORMAT ALSO sample_fmt=AV_SAMPLE_FMT_S16. 
Thanks

leomen3, 2018-09-21:
@iizukanao, excellent tool, thank you for the contribution!
One Issue I have encountered:  when using RPi3 4 microphone HAT (http://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/), is that picam cannot start audio capture:
$ _./picam --alsadev hw:1,0_

> State file open failed: Permission denied
> configuring devices
> error: cannot set channel count for microphone (Invalid argument)

iizukanao, 2018-09-22:
@leomen3 Can you try adding `-c 4` to the command line arguments?
i.e. `$ ./picam --alsadev hw:1,0 -c 4`

leomen3, 2018-09-22:
@iizukanao , thank you for prompt response.
From picam help, I though only 1 or 2 are a valid parameters.
I tried 4 in any case, and got the below:

> error: invalid channels: 4 (must be 1 or 2)
I am using picam version 1.4.7.