iizukanao/picam-streamer #10: WIFI won't work

dogebit, 2017-06-15:
Hi there,
I am trying to set up and run picam-streamer on my pi, but i got stuck with trying to fix the wifi.
I tried multiple times to configure wifi, but nothing seemed to work. I even edited wpa_supplicant, but it just wouldn't connect. Can anyone help me on this?

iizukanao, 2017-06-17:
@dogebit Are you able to connect to your WiFi when you use [the official Raspbian image](https://www.raspberrypi.org/downloads/raspbian/)?

dogebit, 2017-06-17:
I thought I couldn't connect, but then I found that I made a typo with some quotes in wpa_supplicant.
I am now going to close this, thanks for helping!

closed 2017-06-17 by dogebit

iizukanao, 2017-06-18:
That's good to hear. Thanks for using picam-streamer!