iizukanao/picam-streamer #13: index.m3u8 seems to be missing

chtoller, 2017-11-16:
Hi there!

First of all, picam and picam-streamer are really great tools! I'm having lots of fun adding sound-triggered recording. Not there yet, but close :)

But I do have a sudden issue. After moving my Pi (physically), the life stream is gone.
http://192.168.1.10/view/ shows an all white screen. Only the player controls are shown, but hitting the play button doesn't do anything.

Chrome Inspector console says: GET http://192.168.1.10/video/index.m3u8 404 (Not Found)
So it seems that index.m3u8 is missing. Well, /home/pi/public/video/ is certainly empty.

Funnily, if I modify server.js to enable HLS (as you discribed in another issue), I can see the life stream on my mobile via http://192.168.1.10/hls/index.m3u8. So the camera is ok,

Any idea what could be causing this?
Which part of the program creates index.m3u8 and where can I see related errors?

Of course I could re-install the picam-streamer image, but I'm hesitant because setting up my development environment was quite a pain.

Thanks,

Christian
[tethis IT, Austria](https://tethis-it.at)

iizukanao, 2017-11-17:
@chtoller Thank you for using picam! I think Flash Player is not enabled in your browser. Please try enabling Flash Player in Chrome.

Also, you are right. HLS (index.m3u8 and *.ts files) is not enabled by default in picam-streamer, but you can manually enable it by modifying server.js. If HLS is enabled, index.m3u8 will be created automatically.

chtoller, 2017-11-18:
Thanks a lot Iizuka-san, that was exactly the issue.

Shame on me! I tried Chrome and MS Edge before raising the request.
When I tried Firefox yesterday (before I saw your answer), it told me that flash was missing.
Reinstalling Flash then solved the issue for all browsers.

closed 2017-11-18 by chtoller