iizukanao/picam-streamer #14: nohup: redirecting stderr to stdout

TomUser1, 2017-11-21:
Current version (2017-11-09-picam-streamer-1.4.6.zip) not working, link http://RASPI_IP/view/ not working (raspbery 3, zero) and when I run /picam-streamer/script/start.sh result is: 
creating directory structure...
starting node-rtsp-rtmp-server and picam...
nohup: redirecting stderr to stdout

end exit.

  Can you check boot and start script?

iizukanao, 2017-11-22:
@TomUser1 I've confirmed that it works on my Raspberry Pi 3 Model B. That output message including "nohup" is not an error. It's successful. Did you replace *RASPI_IP* with your Raspberry Pi's IP address? Also, you don't need to run start.sh manually.

TomUser1, 2017-11-22:
Raspberry 3 model B works good (have streaming on http and rtsp too), but on raspberry zero makes some errors - By default No stream to http or RTSP protocols and when i start node script (./start_server.sh) manually result is:  "sudo: coffee: command not found".
and when manually run /picam-streamer/script/start.sh,   rusult is:  
"creating directory structure...
starting node-rtsp-rtmp-server and picam...
nohup: redirecting stderr to stdout".
And Where I can find Log File?
Yes I replace RASPI_IP with my Raspberry Pi's IP address.
Thanks

iizukanao, 2017-11-22:
@TomUser1 Sorry, I haven't tested it on Raspberry Pi Zero.
The log messages are suppressed by default, but you can enable them by editing start.sh and changing `/dev/null` at line 31 to some file name such as /home/pi/out.log.

TomUser1, 2017-11-22:
Thanks,
on raspberry pi3 model B connected 2 MEMS microphone via i2s interface. When i Stream audio via vlc it works fine, but when stream via picam it have some noise (with audio). Do it some parameters to be clear?
Thanks

TomUser1, 2017-11-22:
I change simple rate parameter from 48000 to 32000 and - audio have not noise

TomUser1, 2017-11-23:
[vlc-record-.zip](https://github.com/iizukanao/picam-streamer/files/1498291/vlc-record-.zip)
But After about 5 minutes there is a noise.
Recorded file from rtsp stream:

iizukanao, 2017-11-23:
@TomUser1 Hmm, there isn't any noise even after 5 minutes without picam?

TomUser1, 2017-11-26:
NO, when I stream audio from MEMS mic (dmic_sv) via VLC, that works fine without any noise. How do you think in what there can be a problem? 
Thanks

TomUser1, 2017-11-27:
whether there can be a problem in it? :
"$arecord -D mic_sv -r 44100 -f S16_LE -t wav -d 0 -c 2 test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
arecord: set_params:1233: Sample format non available
Available formats:

S32_LE"
Сan you add support of the S32_LE FORMAT ALSO sample_fmt=AV_SAMPLE_FMT_S16.
Thanks

iizukanao, 2017-12-03:
@TomUser1 OK, I will give it a try.

rajeshkrarora1, 2019-10-30:
@TomUser1 
Hi dear, instead of '-D mic_sv', use '-D mic_hw ', there will be no hummimg noise. this suggested by Dave4675 #issue30