iizukanao/picam-streamer #19: Please document that pi-1 and pi-zero will not work with picam-streamer as shipped

Geefer, 2017-11-27:
I was trying to use picam-streamer on a pi zerow and found that it would not start. 

After much messing around I discovered that the node program will not run (gives illegal instruction error) since the version installed on Raspbian Stretch does not support Armv6 architectures as used on the Pi-1 or Pi-zero(w)
It is possible to replace the version installed by downloading directly from nodejs.org a suitable pre-built version and I now have the streamer working (though I still have issues getting it to start correctly from init.d but I think that may be my fault somehow)

It would be great if the picam-streamer documentation made it clear that there was this issue.

iizukanao, 2017-12-03:
@Geefer Thank you for your detailed report! I will be happy to replace the Node.js binary in order to support armv6. Will do it within a few days.

iizukanao, 2017-12-05:
@Geefer Updated the binary for armv6. Please download it from here. http://s.kyu-mu.net/picam-streamer-latest/

Geefer, 2017-12-16:
Thanks. I have downloaded the new image and confirm it runs on the Pi ZeroW

closed 2017-12-16 by Geefer