iizukanao/picam-streamer #24: index.m3u8 not found

briananstett, 2018-06-26:
Hey @iizukanao, Thanks for your contribution. I'm having an issue loading the stream index.
[image]

[image]

mtrojan, 2018-07-05:
Can someone help with this issue?

iizukanao, 2018-07-07:
@briananstett Sorry for the late reply. It shouldn't be a problem as long as you can view the stream. You have to enable Flash Player.

mtrojan, 2018-07-07:
Hi, as zou can see on the screenshot we can.t view the stream. It is blank and the same error 404 index.m3u8 not found.

I have RaspberryPi2, Raspberry camera and USB mic Jabbra 410. I have changed audio rate against the original config. 

I have started with https://kamranicus.com/guides/raspberry-pi-3-baby-monitor
but I have stucked with streaming to VLC. It is seems that is not making any video streem with my config. How can I check? Is picam creating any files I can see it is working? Or any log?

Where should be an issue?

iizukanao, 2018-07-07:
@mtrojan Have you enabled Flash? Flash Player is required when you view your stream with web browser.

mtrojan, 2018-07-09:
Hi, so I have enabled Flash, but it gives me an error 2154. I thing it is related with absence of the index.m3u8. Any idea?

iizukanao, 2018-07-10:
@mtrojan I'm afraid I don't know how to fix it.

LODweb, 2019-09-20:
Hey - just to say - I get the same #2154 error :( 
[image]