iizukanao/picam-streamer #25: Black camera view in the browser

ErHaMep, 2018-09-17:
Hi,

I have installed the picam-streamer.
Now after some other startup problems the picam-streamer will not work.
In the pc browser I see in the browser window a black camera view.

[image]

In issue https://github.com/iizukanao/picam/issues/104 I checked the powersupply: it is OK.
In issue https://github.com/iizukanao/picam-streamer/issues/6 you explain to check the camera. The camera is working well.

I have tried it with Firefox and Edge.
Edge said: Unknown source.
Firefox shows the picture above.

When I run the pc browser the communication leds of thr RPi are flashing faster.
In firefox the newest flash player addon is installed: Shockwave Flash 31.0.0.108.

What is going wrong here?

Roelof

ErHaMep, 2018-10-03:
Hello,

Some more remarks from my site.

The problem can stay on two sides.
It is possible that the pc receives the video, but it is not shown on the monitor or
that RPi and picam-streamer send a black video stream.

In Firefox I use the link: http://RASPI_IP/view/

I have also tried it with VLC, but that shows nothing. Only on lower left side the time that the system is running.
In VLC I use the link: rtsp://RASPI_IP:80/live/picam

RASPI_IP = IP address of the RPi

With putty I can normally access the RPi.
The system load seen in top for picam is about 8 % and after starting Firefox and/or VLC it is about 12 %.

The RPi and the pc are both connected to the local network by a cable and my small laptop by wifi.

In the picture I show the flashplayer used in Firefox, is this one correct?
The player is also enabled.

[image]

Are there more users with the same problem?

iizukanao, 2018-10-06:
@ErHaMep I've updated SD card image and it is now based on the latest Raspbian. Can you give it a try?

ErHaMep, 2018-10-08:
Hi,
The new image works very well. Also the sound is good working.
Thanks very much!
Now I want to change the aspect ratio to 1296 pixels by 972 pixels or 4:3.
I have to find out where picam is started to change the command line options. 
A little difficult for me. Linux is not the same as windows/msdos....
And when the picture is moving the video becomes noisy, I want also look for that.
I will use picam in a birdbox.

closed 2018-11-07 by ErHaMep

jhamilt0n, 2020-07-08:
I too am getting just a blank screen.  Is there a workaround?

iizukanao, 2020-07-16:
@jhamilt0n Please download the updated image. If it does not work, please open a new github issue.