iizukanao/picam-streamer #27: will this image work with raspberry pi zero w, wirelessly

rajeshkrarora1, 2018-10-08:


closed 2018-11-03 by rajeshkrarora1

iizukanao, 2018-11-03:
@rajeshkrarora1 Sorry for this late response. I haven't tested this image with Raspberry Pi Zero W. I'm not sure if it works.

rajeshkrarora1, 2018-11-03:
i have tested it, and thankyou for this wonderful work, as it is working
wonderfully with raspberrypi zero w wirelessly, the only issue i'm facing
is with audio as i have a adafruit i2s mic SPH0645 and i'm still unable to
make it work with Picam-Streamer, though i've installed every thing for
this mic to work, alsamixer shows this module as " snd_rpi_simple_card" and
i'am able to change it's volume in alsamixer but picam-streamer does not
use it to steam audio, i need to understand how to change the audio input
device in picam-streamer, pls help me with this as i need the mic to be as
small as possible and so cannot use usb mic or  Wolfson Audio Card