iizukanao/picam-streamer #30: adafruit I2S mic SPH0645 SUPPORT

rajeshkrarora1, 2018-10-29:
I'am try to use adafruit I2S mic SPH0645 for audio input to PICAM-STREAMER on raspberrypi zero w with V2 official camera with  no success, video stream works perfectly but with no audio though i have tweaked the audio settings as per the instructions at https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/raspberry-pi-wiring-and-test#raspberry-pi-i2s-configuration but with no success

rajeshkrarora1, 2018-11-01:
first of all let me thankyou for this wonderful project it works fine on my pi zero w, but without audio, I downloaded the picam-streamer image from http://s.kyu-mu.net/picam-streamer-latest/ link, added wpa_supplicant and ssh to run it wirelessly,  i have followed the instructions at https://learn.adafruit.com/adafruit-i2s-mems-microphone-breakout/raspberry-pi-wiring-and-test to install SPH0645 i2s mic by adafruit support on my pizero as a root, i have altered values in .asoundrc as per http://www.codingmama.io/tutorials/audio-streaming-with-raspberry-pi-zero-w-and-adafruits-i2s-mic,  https://github.com/iizukanao/picam/issues/86, nothing is working for audio, i stopped "picam", installed vlc, and streamed audio using cvlc as per http://www.codingmama.io/tutorials/audio-streaming-with-raspberry-pi-zero-w-and-adafruits-i2s-mic, it works fine but i'am unable to stream audio and video together, if i use "./picam -w 640 -h 480 --alsadev plug:dmic_sv -r 44100 --verbose --rtspout -v 500000 -f 20 --autoex" command, i get an error "error: failed to set camera to idle state
Perhaps another program is using camera, otherwise you need to reboot this pi" if i kill the "picam" process and then use above command it says "error: failed to connect to video data socket /tmp/node_rtsp_rtmp_videoData): Connection refused
perhaps RTSP server (https://github.com/iizukanao/node-rtsp-rtmp-server) is not                                              running?, actually i do not have a clue how to I install and compile all of the modules needed to use with "picam", cou;ld you please help

rajeshkrarora1, 2018-11-01:
hardware is 
Adafruit I2S MEMS Microphone Breakout - SPH0645LM4H [ADA3421

Dave4675, 2018-11-02:
I have adafruit microphones too, but I'm using them stereo, now the thing is that the standard adafruit asoundrc configuration is not compatible with streaming because it uses S32_LE format by default, so you need to use alsa plug in in order to have a linear conversion, in this way the alsa module will adapt automatically to the need of the stream. Anyway, this is my asoundrc configuration file 

Take care of two things, first, the mono configuration will result into audio non synched, two this issue doesn't mean anything with picam sofware itself

pcm_slave.sl2 {
	pcm "plughw:1,0"
	format S32_LE
	rate 48000
}

pcm.!default {
	type rate
	slave sl2
}

ctl:!default {
	type rate
	slave sl2
}

Dave4675, 2018-11-02:
I don't use the picam streamer, so you also need to configure your alsa input device

rajeshkrarora1, 2018-11-02:
Thanks for a prompt response, i have asoundrc at ~/.asoundrc, and without picam-streamer only audio streaming was working fine with asoundrc looking like this
pcm.dmic_hw {
  type hw
  card sndrpisimplecar
  channels 2
  format S32_LE
}
pcm.dmic_sv {
  type softvol
  slave.pcm dmic_hw
  control {
    name "Boost Capture Volume"
    card sndrpisimplecar
  }
  min_dB -2.0
  max_dB 10.0
}

rajeshkrarora1, 2018-11-02:
i have not understood where to place .asoundrc in picam-streamer so that that picam-streamer uses the adafruit i2s mems mic and streams audio/video together

rajeshkrarora1, 2018-11-02:
i have made my asoundrc look like yours and rebooted with no success, could you please suggest where should i place my .asoundrc

Dave4675, 2018-11-03:
the ~/.asoundrc is a configuration file used by the alsa driver, it is located into the home directory. I don't know how to confugure picam-streamer because I use the picam software, not the streamer. but it should work, try to open an issue referring about how to select the input device.

rajeshkrarora1, 2018-11-03:
Thanks Dave4675 for this clue, god bless