iizukanao/picam-streamer #34: Router adjustments using picam-streamer

ErHaMep, 2019-01-02:
Hi, 

In the documentation of picam I read nothing about the adjustment of the router when you want to use picam outside the local network.
In the file ~/node-rtsp-trmp-server/config.js I see many adjustmets for ports:

    // Server listen port
    serverPort: 80,
    // RTMP server listen port
    rtmpServerPort: 1935,

    // For receiverType == 'tcp' or 'udp'
    receiverListenHost: '0.0.0.0',
    videoControlReceiverPort: 1111,
    audioControlReceiverPort: 1112,
    videoDataReceiverPort: 1113,
    audioDataReceiverPort: 1114,

    /* RTSP configurations */
    // Server ports for RTP and RTCP
    audioRTPServerPort: 7042, // even
    audioRTCPServerPort: 7043, // odd and contiguous
    videoRTPServerPort: 7044, // even
    videoRTCPServerPort: 7045, // odd and contiguous

    // UDP port numbers to receive incoming RTP data
    rtspVideoDataUDPListenPort: 5004,
    rtspVideoControlUDPListenPort: 5005,
    rtspAudioDataUDPListenPort: 5006,
    rtspAudioControlUDPListenPort: 5007

Is it neccesary to open all this ports in te router to use picam externally?

ErHaMep

iizukanao, 2019-01-03:
It depends on your use case.
- `serverPort`: For HTTP and RTSP
- `rtmpServerPort`: For RTMP
- `videoControlReceiverPort`, `audioControlReceiverPort`, `videoDataReceiverPort`, `audioDataReceiverPort`: Only necessary when you use picam with `--rtspout` option
- `audioRTPServerPort`, `audioRTCPServerPort`, `videoRTPServerPort`, `videoRTCPServerPort`, `rtspVideoDataUDPListenPort`, `rtspVideoControlUDPListenPort`, `rtspAudioDataUDPListenPort`, `rtspAudioControlUDPListenPort`: For RTSP

ErHaMep, 2019-01-05:
Thanks, I will try it.
ErHaMep

closed 2019-01-05 by ErHaMep