iizukanao/picam-streamer #36: how to change the http link to my custom link

udgish23, 2019-01-31:
hi I want to change the http://PI_ip/view/ to something like http://camera2/view/ where can i config that I didn't find the settings

iizukanao, 2019-01-31:
@udgish23 If you want to use a public URL, please use some Dynamic DNS service to bind your IP address to a domain. If you just want to use some local machine instead of an IP address, please edit your PC's *hosts* file.

udgish23, 2019-02-02:
Thanks for reply ! could you please describe that how can we transfer recorded file and messages on other device using this example and also could you please give some idea about Dynamic DNS service.

iizukanao, 2019-02-03:
You can use `scp` command to transfer files from Raspberry Pi to your computer.

Dynamic DNS is only necessary if you want to access your Raspberry Pi over the internet. I think [Duck DNS](https://www.duckdns.org/) will work for you.

udgish23, 2019-05-10:
I want to set audio and video quality proper in  very dim light like night mode.

  Streaming voice is clear and audible but recorded voice is not clear can you suggest any changes or command we can do.

Can we set shutter speed of camera and how to save changes made like exposure and white balance?