iizukanao/picam-streamer #40: Instructions?

jpt000, 2019-05-20:
Picam-server works great out of the box. Nevertheless it would be nice to have a little more information, namely on how to change settings. Took me a couple of hours to figure out where to look:
/home/pi/picam-streamer/lib/server.js 
Thanks for the hard work! Very much appreciated!

kerwinf, 2020-02-15:
I am starting to use this for a RPi0w and am curious how to set default values at launch. My camera seems to have too much red so all the colors are off. Everything is pinkish purple.

iizukanao, 2020-02-16:
@jpt000 Sorry, I missed your message!

@kerwinf You might want to set white balance using picam's `--wb <value>` option. You can add picam options by editing line 44 of /home/pi/picam-streamer/lib/server.js like this:

```
    picam = child_process.spawn('nice', ['-n', '-20', 'ionice', '-c1', '-n0', 'sudo', '-u', "#" + PICAM_UID, PICAM_PATH, '--rtspout', '--alsadev', 'hw:1,0', '--wb', 'fluorescent' ], {
```

iizukanao, 2020-02-16:
See https://github.com/iizukanao/picam#command-options for the list of picam options.

TomTomCountry, 2021-11-03:
You have done a really great job. I finally got my picam running after trying for some weeks in my freetime. But I still struggle to rotate the picture. I have added the '--rotation 180' to the server.js file, but it seems not working. I even tried to put it in brackets <>.
So I will be thankful for any tip ;-)
And maybe some very basics for the most, but how can I restart picam-streamer via ssh without rebooting?
Thanks in advandce

iizukanao, 2021-11-03:
@TomTomCountry Hi, thanks for using picam-streamer!
`sudo service picam-streamer restart` will do the job!

iizukanao, 2021-11-03:
As for rotation, could you share the line you have edited?

TomTomCountry, 2021-11-04:
Hi @iizukanao sure,
here is the line for the rotation
`picam = child_process.spawn('nice', ['-n', '-20', 'ionice', '-c1', '-n0', 'sudo', '-u', "#" + PICAM_UID, PICAM_PATH, '--rtspout', '--alsadev' , '--rotation 180', 'hw:1,0'], {
`
Thanks for your fast response :)

iizukanao, 2021-11-05:
@TomTomCountry You have to write `'--rotation', '180'` instead of `'--rotation 180'`. Also, there must be `'hw:1,0'` right after `'--alsadev'`, so put `'--rotation', '180'` at the very end of the arguments.
```js
picam = child_process.spawn('nice', ['-n', '-20', 'ionice', '-c1', '-n0', 'sudo', '-u', "#" + PICAM_UID, PICAM_PATH, '--rtspout', '--alsadev', 'hw:1,0', '--rotation', '180'], {
```

commit cb6ea85f (2021-11-05): Describe how to configure audio/video settings

Fixes #40
closed 2021-11-05 by iizukanao