iizukanao/picam-streamer #43: Change Audio Device

Medium999, 2020-09-02:
Hi Iizukanao, Picam streamer is great! There seem to be so many programs for streaming video but hardly any with an Audio option.
I am very new to this but I managed to get the Picam stream working following your instructions. The only part i can't get working is the Audio stream. 
I am using a web cam with a mic for the audio. I am able to open it in alsamixer and adjust the input level as per your instructions but how do i configure the webcam mic to be the main device for recording through to the stream?
arecord -l tells me my USB Audio devicde is on card 3 device 0 so maybe the default is on a different card and i have to change it?

Medium999, 2020-10-11:
Can anyone help me? It would be really appreciated.

ErHaMep, 2020-10-12:
Hi Medium999,

I have also a problem with the microphone and picam-streamer version 1.4.9.
But in version 1.4.7 it is all OK. I used it for a few years.
This evening I have tried new setups with 1.4.7 and 1.4.9 and with 1.4.7 it works and with 1.4.9 not.
1.4.9 is also made with a new version of Raspberry Pi OS.
So the problem can exist in the picam-streamer or in the Pi OS. I think in the Pi OS it goes wrong.
In the picam part of Nao Iizuka's github you can read information about the audio device.
It seems you can made adjustments in some files to set the correct hardware numbers for picam-streamer.
During the next days I will try this.

I use picam-streamer in a birdbox andit works very well.

Best regards,
ErHaMep

ErHaMep, 2020-10-14:
Hi
I have fixed the problem and may be also your problem.
The problem is that the new os or the different types of Raspberry has more reserved card numbers for the audio devices.
In picam-streamer for the mic card number 1 is used, but card 1 is also reserved by the system. The inserted USB mic becomes number 2 or 3 in your case..

For your information read [https://github.com/iizukanao/picam-streamer/issues/3](url) and find the picam option --alsadev in line 44. The mic card number can be set in the picam system by changing option '--alsadev', 'hw:1,0' to '--alsadev', 'hw:?,0', ? is the card number.

You can find the current card number with the command $ arecord -l
In my rpi it shows:
**** List of CAPTURE Hardware Devices ****
card 2: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

I have changed 'hw:1,0' to 'hw:2,0' and the mic worked after a reboot of the rpi.
You have to change it to 'hw:3,0'

May be works this solution also for you.

I include also the picam doc file.
[picam-help.txt](https://github.com/iizukanao/picam-streamer/files/5380817/picam-help.txt)

ErHaMep

Medium999, 2020-11-15:
Thanks @ErHaMep for your reply!
I finally got around to testing your suggested solution. 

I used --alasadev hw:2:0

The result is as follows

configuring devices
error:failed to set the camer to idle state
Perhaps another program is using camera, otherwise you need to reboot this pi

On reboot i get the same the same error 

I assume i have to somehow stop the camera?

I looked online to try and find v1.4.7 but can not find it, do you have a copy?

ErHaMep, 2020-11-17:
Hi,

The message that the camera is used looks like that you try to start picam manually.
The "out of the box SD image" starts automaticly  the camera and streamer at power up.
I have the message also seen in the past and concluded that the camera was already running.
On this moment when I start $./picam I get the same message as you.

You can test the running of picam with the $ top command.
It shows all the running processes in the pi.
The most busy tasks are at the top of the list in this case picam.

What also is possible that you have made a mistake in changing the file server.js

I have a copy of v1.4.7 and you can get it.
But may be it works not with the Raspberry Pi 4
You download it with the link below from wetranfer.
After 24 November 2020 it is automaticly deleted.
https://we.tl/t-cxotCa0gu2

In your message from 2 Sep you said that the audio device is on card 3.
In the last message you say: I used --alsadev hw2:0.
Is there something changed? hw2 is card 2.

Manual start of picam
[image]

Task list
[image]