iizukanao/picam-streamer #44: http stream not working on chrome

rajeshkrarora1, 2020-11-06:
http stream stopped  working on chrome though it works on microsoft-edge, as we know all browsers are stopping the ability to run Adobe-flash, is there any work around

Medium999, 2020-11-15:
This worked for me, for chrome on PC you can specifically enable flash for the URL you are visiting

https://www.umass.edu/it/support/support-center/howenableflashyourwebbrowser#Google%20Chrome

However, it would be great if PiCam Streamer didn't use flash, as you mentioned it is outdated & can't use it on my Android phone.

rajeshkrarora1, 2020-11-15:
@Medium999
Thanks for your reply, it started working after some time on chrome that
day, but the issue is that after 31dec2020, adobe will stop flash support,
i.e no updates etc and all the browsers are also stopping support and
ability to use flash through updates