iizukanao/picam-streamer #45: how to change directory

rajeshkrarora1, 2021-03-26:
Hi sir, Thankyou for this wonderful project.
I need to change the recording directory from picam-streamer/archive to a directory in the attached usb drive, I want to directly record the videos on a attached USB pendrive, please help me with that

rajeshkrarora1, 2021-04-02:
Sir, I'am using ur script in "https://github.com/iizukanao/picam/issues/89" for recording, is it possible to write a script to check and 'mv' all the video files in "picam-streamer/archive", viz "picam-streamer/archive/*.ts "  to a folder in mounted USB drive to raspberry pi continuously as and so that also maintain a circular buffer of recording

iizukanao, 2021-04-03:
Hi @rajeshkrarora1,
Apologies for the delayed response. Yes, you can replace the *archive* directory with a symbolic link in order to store recorded files into somewhere else. If you mount a USB drive at */mnt/mydisk*, run these commands:

```sh
sudo service picam-streamer stop
mv /home/pi/picam-streamer/archive/* /mnt/mydisk/
rmdir /home/pi/picam-streamer/archive
ln -s /mnt/mydisk /home/pi/picam-streamer/archive
sudo service picam-streamer start
```

rajeshkrarora1, 2021-04-03:
Hi Sir, Thank you for your response. This is working like magic now

iizukanao, 2021-04-03:
Great, thanks for the feedback! 🙂

closed 2021-04-03 by iizukanao