iizukanao/picam-streamer #46: picam-streamer has stopped working after upgrading to node.js version 12 or 14

rajeshkrarora1, 2021-08-02:
Hi Sir, I was using node-red with picam-streamer, it was working wonderfully up till now , i had to upgrade node.js as node-red dropped support to  node.js  version 10 and below, dashboard nodes after upgrading the stream at the time of boot stopped completely, i tried to start it by giving command "sudo /etc/init.d/picam-streamer start", it gives an out put,

Starting picam-streamer
creating directory structure...
starting node-rtsp-rtmp-server and picam...
nohup: redirecting stderr to stdout
nohup: failed to run command '/usr/local/bin/node': No such file or directory.

iizukanao, 2021-08-02:
What is the output of the command `which node`?

rajeshkrarora1, 2021-08-02:
Thankyou for responding so promptly

it pi@raspberrypi:~ $  which node
/usr/bin/node

rajeshkrarora1, 2021-08-02:
🙏🙏

rajeshkrarora1, 2021-08-02:
how should i solve this problem, where do in need to edit

iizukanao, 2021-08-02:
You can just run `sudo ln -s /usr/bin/node /usr/local/bin/node` and the problem will be fixed.

rajeshkrarora1, 2021-08-02:
Thankyou so so very much, for your support 🙏🙏🙏🙏

iizukanao, 2021-08-02:
You're welcome 🙂

closed 2021-08-02 by iizukanao