iizukanao/picam-streamer #47: Can RTMP stream be password protected

rajeshkrarora1, 2021-08-09:
Hi iizukanao, 
I wanted to know if the RTMP stream be password protected, i being viewed on VLC

iizukanao, 2021-08-09:
As for "password protected", do you want a Basic HTTP authentication?
Also, regarding "stream.key", does it mean you would like to use a URL of your choice instead of the fixed one?

rajeshkrarora1, 2021-08-09:
Hi, iizukanao, I use ngrok tcp forwarding for the RTMP stream to watch it
on VLC, i have seen some camera RTSP streams which are username:password
protected while watching on VLC, .

If some thing similar can be implemented on the RTMP stream from Picam ,
because RTMP stream from Picam is consistent.
 As we use Encrypted Key on youtube, if some thing similar can be used
while watching on VLC, from security point of view.

iizukanao, 2021-08-10:
Unfortunately, that is not supported in picam-streamer at this time. As an alternative, you can secure your pi by setting up a firewall using ufw so that only some source IP addresses are allowed to access the stream.

rajeshkrarora1, 2021-08-10:
Thankyou for your prompt Reply🙏🙏🙏🙏