iizukanao/picam-streamer #48: Can picam-streamer be run on any other ARM board eg NVIDIA JETSON

rajeshkrarora1, 2021-08-09:
Hi iizukanao, Are you planning to make picam-streamer work on any other ARM board capable of H.265 encoding, the draw back of using it on raspberry pi is it's video encoding H.264 only

iizukanao, 2021-08-09:
Unfortunately, I don't have a plan for supporting devices other than Raspberry Pi.

rajeshkrarora1, 2021-08-09:
Thankyou for your early reply, Raspberry pi is a very versatile  board, but
there are few limitations specially the video encoder only supports H.264,
so if we want to stream 4K at better FPS, then we are stuck.🙏🙏🙏🙏

iizukanao, 2021-08-10:
Understandable. Is there any alternative to Raspberry Pi that comes with H.265 hardware encoding?

rajeshkrarora1, 2021-08-10:
Hi iizukanao, The  Odroid C2  and Rock64   support H.265

iizukanao, 2021-08-10:
Thanks. If I have a chance to use one of the boards, I'll take a look.

rajeshkrarora1, 2021-08-10:
Thankyou 🙏🙏🙏🙏